DC-SWAT/DreamShell #4: [Feature] Add new JC-BootROM 1.032 BIOS to DreamShell/build/firmware/bios/sega/

Keizel, 2017-05-12:
Please, could you add the new Bios and compilate it for the web? This way, the people that download in DC-Swat the .cdi image could install the new Bios without SD reader.

Link for download the new BIOS: http://www.mediafire.com/file/71bvzm5e1bnzcb5/jc-bootROM-v1.032.zip

New about new BIOS: http://japanese-cake.livejournal.com/10056.html

megavolt85, 2017-05-12:
JC-BootROM is not part DS you can use Lazzy Boot for rebuild inage with any files

closed 2017-05-12 by megavolt85

Keizel, 2017-05-12:
@megavolt85 but JC-BootROM IS in build/firmware/bios/sega but the version that the folder contain is old (1.031). The new version is 1.032. I did pull request for change it to the lastest version: https://github.com/DC-SWAT/DreamShell/pull/5

If DreamShell have 1.031, I say that is better have the lastest version (1.032), right? You can see the old version inside DreamShell in this folder: https://github.com/DC-SWAT/DreamShell/tree/master/build/firmware/bios/sega

For this, I did pull request for update JC-BootROM.