DC-SWAT/DreamShell #6: Can't build DreamShell

Nold360, 2018-01-25:
Hi,

I'm trying to create a Docker-Container to build DreamShell within. 
I already created a container including the KOS-Toolchain, which works just fine.

Based on that container, I build the DS-Toolchain & libs, patched & rebuild KOS. Every Makefile exits with Returncode 0. See the Dockerfile for details. [0]
I forked DS to fix KOS-Patching & change KOS-Location in Makefile. See [1] for details.

But I still got a problem linking DS:
```
/opt/toolchains/dc/kos/utils/genexports/genexports.sh exports.txt ./src/exports.c ds_symtab
/opt/toolchains/dc/kos/utils/genexports/genexportstubs.sh exports.txt ./src/exports_stubs.c
make -f Makefile.stubs
make[1]: Entering directory '/opt/toolchains/dc/kos/ds'
kos-cc  -c src/exports_stubs.c -o src/exports_stubs.o
rm -f ./sdk/lib/libds.a
/opt/toolchains/dc/sh-elf/bin/sh-elf-ar rcs ./sdk/lib/libds.a ./src/exports_stubs.o
make[1]: Leaving directory '/opt/toolchains/dc/kos/ds'
kos-cc  -c src/exports.c -o src/exports.o
/opt/toolchains/dc/kos/utils/genexports/genexports.sh exports_gcc.txt ./src/exports_gcc.c gcc_symtab
/opt/toolchains/dc/kos/utils/genexports/genexportstubs.sh exports_gcc.txt ./src/exports_gcc_stubs.c
make -f Makefile.gcc_stubs
make[1]: Entering directory '/opt/toolchains/dc/kos/ds'
kos-cc  -c src/exports_gcc_stubs.c -o src/exports_gcc_stubs.o
rm -f ./sdk/lib/libds_gcc.a
/opt/toolchains/dc/sh-elf/bin/sh-elf-ar rcs ./sdk/lib/libds_gcc.a ./src/exports_gcc_stubs.o
make[1]: Leaving directory '/opt/toolchains/dc/kos/ds'
kos-cc  -c src/exports_gcc.c -o src/exports_gcc.o
/opt/toolchains/dc/kos/utils/genromfs/genromfs -f romdisk.img -d romdisk -v
0    rom 5a699a15         [0xffffffff, 0xffffffff] 37777777777, sz     0, at 0x0
1    .                    [0xc4      , 0x97ca6c  ] 0040755, sz     0, at 0x20
1    ..                   [0xc4      , 0x97c281  ] 0040755, sz     0, at 0x40     [link to 0x20    ]
1    logo.kmg.gz          [0xfd18    , 0x4a958f  ] 0100644, sz 31403, at 0x60
/opt/toolchains/dc/kos/utils/bin2o/bin2o romdisk.img romdisk romdisk.o
/opt/toolchains/dc/sh-elf/bin/sh-elf-gcc -O2 -fomit-frame-pointer -ml -m4-single-only -ffunction-sections -fdata-sections  -I/opt/toolchains/dc/kos/include -I/opt/toolchains/dc/kos/kernel/arch/dreamcast/include -I/opt/toolchains/dc/kos/addons/include -I/opt/toolchains/dc/kos/../kos-ports/include -D_arch_dreamcast -D_arch_sub_pristine -Wall -g -fno-builtin -fno-strict-aliasing -I/opt/toolchains/dc/kos/ds/sdk/include -I/opt/toolchains/dc/kos/ds/sdk/include/SDL -DTARGET_NAME= -DDS_ARCH_DC -I./include -I./include/SDL -I./include/fatfs -I./include/ntfs -DHAVE_SDLIMAGE -DVER_MAJOR=4 -DVER_MINOR=0 -DVER_MICRO=0 -DVER_BUILD=0x24    -ml -m4-single-only -Wl,-Ttext=0x8c010000 -Wl,--gc-sections -T/opt/toolchains/dc/kos/utils/ldscripts/shlelf.xc -nodefaultlibs -L/opt/toolchains/dc/kos/lib/dreamcast -L/opt/toolchains/dc/kos/addons/lib/dreamcast -L/opt/toolchains/dc/kos/../kos-ports/lib -L./lib -o DS.elf \
        ./src/main.o ./src/video.o ./src/console.o ./src/gui/gui.o ./src/commands.o ./src/module.o ./src/events.o ./src/fs/fs.o ./src/lua/lua.o ./src/lua/lua_ds.o ./src/lua/packlib.o ./src/app/app.o ./src/app/load.o ./src/list.o ./src/img/pvr.o ./src/cmd_elf.o ./src/vmu/vmu.o ./src/irq/exceptions.o ./src/irq/setjmp.o ./src/settings.o ./src/drivers/spi.o ./src/drivers/sd.o ./src/drivers/enc28j60.o ./src/drivers/asic.o ./src/drivers/rtc.o ./lib/SDL_gui/SDL_gui.o ./lib/SDL_gui/Exception.o ./lib/SDL_gui/Object.o ./lib/SDL_gui/Surface.o ./lib/SDL_gui/Font.o ./lib/SDL_gui/Callback.o ./lib/SDL_gui/Drawable.o ./lib/SDL_gui/Screen.o ./lib/SDL_gui/Widget.o ./lib/SDL_gui/Container.o ./lib/SDL_gui/FastFont.o ./lib/SDL_gui/TrueTypeFont.o ./lib/SDL_gui/Layout.o ./lib/SDL_gui/Panel.o ./lib/SDL_gui/CardStack.o ./lib/SDL_gui/FastLabel.o ./lib/SDL_gui/Label.o ./lib/SDL_gui/Picture.o ./lib/SDL_gui/TextEntry.o ./lib/SDL_gui/AbstractButton.o ./lib/SDL_gui/Button.o ./lib/SDL_gui/ToggleButton.o ./lib/SDL_gui/ProgressBar.o ./lib/SDL_gui/ScrollBar.o ./lib/SDL_gui/AbstractTable.o ./lib/SDL_gui/ListBox.o ./lib/SDL_gui/VBoxLayout.o ./lib/SDL_gui/RealScreen.o ./lib/SDL_gui/ScrollPanel.o ./lib/SDL_gui/RTF.o ./lib/SDL_gui/Window.o ./lib/SDL_gui/FileManager.o ./lib/SDL_Console/src/SDL_console.o ./lib/SDL_Console/src/DT_drawtext.o ./lib/SDL_Console/src/internal.o ./src/utils.o ./src/utils/gmtime.o ./src/utils/strftime.o ./src/utils/debug_console.o ./src/utils/memcpy.op ./src/utils/memset.op ./src/utils/memmove.op ./src/fs/fat/utils.o ./src/fs/fat/option/ccsbcs.o ./src/fs/fat/option/syscall.o ./src/fs/fat/ff.o ./src/fs/fat/dc.o ./src/exports.o ./src/exports_gcc.o romdisk.o   -lcfg -lmxml -lparallax -lSDL_1.2.13 -lSDL_image_1.2.12 -lSDL_ttf_2.0.11 -lSDL_rtf_0.1.1 -lSDL_gfx_2.0.25 -lfreetype -lkmg -ljpeg -lpng -lz -llua_5.1.4-2 -lkosext2fs -lkosutils -lstdc++ -lm -Wl,--start-group -lkallisti -lc -lgcc -Wl,--end-group
/opt/toolchains/dc/sh-elf/lib/gcc/sh-elf/7.1.0/../../../../sh-elf/lib/libstdc++.a(eh_throw.o): In function `__gxx_exception_cleanup':
/opt/toolchains/dc/kos/ds/sdk/toolchain/build-gcc-sh-elf-7.1.0/sh-elf/libstdc++-v3/libsupc++/../../../../gcc-7.1.0/libstdc++-v3/libsupc++/eh_throw.cc:44: undefined reference to `__gnu_cxx::__exchange_and_add(int volatile*, int)'
collect2: error: ld returned 1 exit status
Makefile:141: recipe for target 'DS.elf' failed
make: *** [DS.elf] Error 1
The command '/bin/sh -c bash -c 'source /opt/toolchains/dc/kos/environ.sh; make'' returned a non-zero code: 2
```

Any advice?

[0] https://github.com/Nold360/docker-kallistios-sdk/blob/dreamshell/Dockerfile
[1] https://github.com/DC-SWAT/DreamShell/compare/master...Nold360:master

megavolt85, 2018-01-25:
use gcc-5.2.0
gcc-7.1.0 have issue with C++

DC-SWAT, 2018-01-25:
Yes, use GCC 5.2.0

closed 2018-01-25 by DC-SWAT

Nold360, 2018-01-26:
Now it's working, thanks a lot! :)