DC-SWAT/DreamShell #7: Failing to compile toolchain on macOS High Sierra

einsteinx2, 2018-05-15:
I followed all of the instructions in the toolchain Makefile (installed gcc 8.1.0 and exported path correctly).

Patching fails on one file for some reason, one of the KOS headers but I was able to easily make the patch changes manually.

Then running `make build` compiles for a while then I hit this error:

```
../../../gcc-7.1.0/libgcc/libgcc2.c \
	  -fexceptions -fnon-call-exceptions -fvisibility=hidden -DHIDE_EXPORTS
../../../../gcc-7.1.0/libgcc/libgcc2.c: In function ‘__divdi3’:
../../../../gcc-7.1.0/libgcc/libgcc2.c:1091:9: internal compiler error: Segmentation fault: 11
      d0 = 1 / d0; /* Divide intentionally by zero.  */
      ~~~^~~~~~~~
libbacktrace could not find executable to open
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
make[5]: *** [_divdi3.o] Error 1
make[4]: *** [multi-do] Error 1
make[3]: *** [all-multi] Error 2
make[2]: *** [all-target-libgcc] Error 2
make[1]: *** [all] Error 2
make: *** [build-sh4-gcc-pass1] Error 1
```

Any ideas? I see that the macOS toolchain build instructions were literally added days ago by @yevgeniy-logachev, so I assume it should work.

einsteinx2, 2018-05-15:
If there's any other info that can help diagnose this, please let me know. Before attempting this, I successfully compiled the KOS toolchain following these instructions: http://dcemulation.org/?title=Compiling_KOS_on_OS_X

Then I moved on to trying to compile DreamShell, but saw that it required it's own toolchain and modified KOS.

einsteinx2, 2018-05-16:
Actually, is the sdk/toolchain folder necessary at all other than the patches to KOS? Since I couldn't get it building, I went ahead and just ran make on KOS again with the patches applied, then ran make on Dreamshell and boom it compiled.

DC-SWAT, 2018-05-18:
Hello.
Don't use gcc 7.1 it's not working properly. Use gcc 5.2.

DC-SWAT, 2018-05-21:
Your problem has been resolved?

einsteinx2, 2018-05-22:
Hey sorry for the late response. Yep I was able to compile DreamShell without a problem using the default GCC 4 based KOS toolchain. I only needed to first apply the included KOS patches (one of them manually because it fails for some reason) and rebuild KOS.

It took a bit more work to get the cdi image building, so here are all of my build notes for anyone else that might want them (feel free to add as a readme to the repo, I think it would be helpful):

```
DreamShell macOS High Sierra notes:

1. First install KOS using the standard instructions from http://dcemulation.org/?title=Compiling_KOS_on_OS_X

2. Clone DreamShell into /opt/toolchain/dc/kos/ds from here: https://github.com/DC-SWAT/DreamShell
   a. Note: must use that path or you need to edit makefiles

3. In the DreamShell sdk/toolchain directory, run "make patch-los"
   a. Note that one of the patches will fail (cdefs.h), but you can manually apply it. 
      To do so, read the diff to find the changes using this command:
      cat patches/kos-a0ba3f8780.diff | colordiff | less -RS

4. Rebuild KOS with the patches applied: cd /opt/toolchains/dc/kos && make

5. Build DreamShell: cd /opt/toolchains/dc/kos/ds && make

6. To build a cdi image of DreamShell, you'll need to rebuild /opt/toolchains/dc/kos/ds/sdk/bin/scramble.exe and /opt/toolchains/dc/kos/ds/sdk/bin/cdi4dc.exe for macOS
   a. Rebuild scramble: cd /opt/toolchains/dc/kos/ds/sdk/bin/src/scramble && make && cp scramble ../../
   b. Rebuild cdi4dc (we can't use the included source so we need to clone a fork and build it): 
      1) cd /opt/toolchains/dc/kos/ds/sdk/bin/src && mv img4dc img4dc.orig
      2) git clone git@github.com:einsteinx2/img4dc.git
      3) cd img4dc && cmake . && make
      4) cp cdi4dc/cdi4dc ../../
   c. Make the cdi image: cd /opt/toolchains/dc/kos/ds && make cdi
```

closed 2018-05-22 by einsteinx2

einsteinx2, 2019-07-27:
For anyone stumbling across this, I'm pretty sure the instructions above no longer work with the latest code from the KOS and Dreamshell repos.

However, even better, I've made a Docker container that contains everything you need to build a Dreamshell cdi image on any machine with Docker installed.

First, install Docker then clone the Dreamshell repo. Pull the docker image with `docker pull einsteinx2/dcdev-dreamshell`. Then from the Dreamshell repo's root directory, launch the container with `docker run -it --rm -v "$PWD:/src" einsteinx2/dcdev-dreamshell:1.0.0` and run the build preparation script with `prep_build` (only needs to be done the first time). After that you can build a cdi image with `make && make cdi`. The image will be in the root directory of the Dreamshell repo on your system.

You can edit the code on the host system, then use the container to build new images for testing.

More info is in the GitHub repo here: https://github.com/einsteinx2/docker-dcdev-dreamshell