DC-SWAT/DreamShell #9: set5v1.001.bin GD-ROM BUG

for2nato, 2019-05-26:
I recently flashed this version to my DC and I lost the use of my GD-ROM. Any disk I put in just hangs on the initial read. If I boot the DC with the disk in the drive it hangs on boot logo. V01 DC was running perfectly with region free bios chip on JC and DS bios before this. Without the use of the GD-ROM I'm stuck on this bios with no way to load DS to flash a good firmware.

megavolt85, 2019-05-26:
this bios for katana devkit, why you flash this bios to retail console?

DC-SWAT, 2019-05-27:
It's not a bug, this BIOS for KATANA Dev Kit, not for retail console. I will remove it from DS to avoid such situations.
Switch to original BIOS then boot from CD and flash another BIOS.

closed 2019-05-27 by DC-SWAT

for2nato, 2019-05-27:
I flashed it because it wasn't labeled as the katana bios and it was in a
package you put together for the dreamcast. It's not unrecoverable but it
is a little bit of a pain in the ass, only because I don't use a dual boot
system. I do full swaps on my bios mods.

Thanks for the response and I think your right to remove it for future
updates.