DC-SWAT/DreamShell #19: Where's VMU Manager App?

ABOhiccups, 2020-07-04:
I downloaded a **build** folder from this GitHub page and renamed **build** folder to **DS** folder and place it to root of SD Card. After boot to DreamShell. VMU Manager App is not there anymore!

Can you please add VMU Manager back in next build version on GitHub? Or you're just waiting until the next major update version on [main website](http://www.dc-swat.ru/page/dreamshell/)?

calumcraig, 2020-08-19:
I have the same issue with the image I downloaded from 

http://www.dc-swat.ru/download/dc/ds/4.0/DreamShell_4.0.0_RC4_CDI_Image.7z

I can launch the image fine using GDEMU but there is no VMU Manager app in the Dreamshell desktop. Am I missing something, should this work? Do I need to use the image with a boot loader?

calumcraig, 2020-08-19:
I should note that I don't currently have an SD card reader attached to the Dreamcast. Maybe VMU Manager is only seen when there is an SD card present?

DC-SWAT, 2020-09-02:
As I known the VMU Manager is broken in latest DS.

calumcraig, 2020-09-02:
Apologies, I should have updated my comment. I realised I was actually using the image for RC3 - when I used RC04 instead VMU Manager is there OK.

DC-SWAT, 2020-11-02:
Now it in build directory.

closed 2020-11-02 by DC-SWAT