DC-SWAT/DreamShell #21: Trying to build on debian, getting errors about gcc

Feathered-Serpent, 2022-11-01:
Heya,

not sure if anyone even still reads this. I tried to compile everything as written at https://github.com/DC-SWAT/DreamShell

Problem 1: it wasn't written, that I need the lua package, which I have installed (maybe add it in the first two apt-gets?)

Problem 2: I'm at the point of the list where I should make source ./environ.sh, but:
[image]

I have downloaded and extracted the gcc 5.2.0 from the webpage, but inside are only Windows executables? How can I compile with them?

DC-SWAT, 2022-11-02:
Hi. Look for [README]([README.md](https://github.com/DC-SWAT/DreamShell#readme)) it has full instruction to setup environment and build toolchain, KOS and DS on any UNIX-like system.

Feathered-Serpent, 2022-11-02:
If you mean https://github.com/DC-SWAT/DreamShell/blob/master/README.md then that's the list of commands I followed. There is no mentioning of downloading or building GCC. Or did I missed something?

megavolt85, 2022-11-02:
steps
```
./download.sh && ./unpack.sh
make && cd ../../
```
it's download and build GCC

Feathered-Serpent, 2022-11-02:
> steps
> 
> ```
> ./download.sh && ./unpack.sh
> make && cd ../../
> ```
> 
> it's download and build GCC

but I'm getting the error two lines above that:
```
source ./environ.sh
cd utils/dc-chain && cp config.mk.testing.sample config.mk
./download.sh && ./unpack.sh
```
source ./environ.sh is what's causing the error in my first post. Or should that error happen and one should just continue?

DC-SWAT, 2022-11-03:
Seems you are missed this step:

`cp ds/sdk/toolchain/environ.sh environ.sh`

DC-SWAT, 2022-11-03:
Try `source ./environ.sh` after toolchain build

DC-SWAT, 2022-11-03:
I fixed instruction please try again:

https://github.com/DC-SWAT/DreamShell/commit/9a65b1943395bf63353f91560d9a47bddf7a7cf0

Feathered-Serpent, 2022-11-03:
> Try `source ./environ.sh` after toolchain build, maybe I should change it in README.
> 
> ```
> cd utils/dc-chain && cp config.mk.testing.sample config.mk
> ./download.sh && ./unpack.sh
> make && cd ../../
> source ./environ.sh
> ```

With this I was able to advance! But now I'm stuck at **cd ../../kos/ds/sdk/bin/src && make && make install && cd ../../../**:
[image]

It seem to not being able to find a file. in the ciso directory there is only this:
[image]

DC-SWAT, 2022-11-03:
Looks like you should install liblzo2-dev.

Feathered-Serpent, 2022-11-03:
Definitely came much farther with libzlo2-dev installed. Besides I also had to install package g++ (I use a vanilla debian system), otherwise I also couldn't build much, same for package lua.

But:
[image]
It is at this point for over two hours now. Why the ip address 192.168.1.110? That is not the address of the debian system, nor any in my network. It's fully utilizing one core since I entered the command.

Feathered-Serpent, 2022-11-03:
Sorry, my fault. I thought, make would have to compile something again, but it's just for starting things!

So I think I finally come to the end. I could run make cdi to get the image. now I'm going to test, if it runs better with Sonic Adventure 2 on my Dreamcast. And if not, I at least learned something about compiling :)

DC-SWAT, 2022-11-05:
> Silly question: what do I do with the CDI file? burning it onto a disc and putting it into the Dreamcast did nothing. the DC only offers me the music player for that disc. Guess there's more than just having the CDI file then.

This is another story, not about DS building :) It's about burning CD's for Dreamcast. Look on forums or something.
If you have SD or IDE device, just put content of the build directory and DS_CORE.BIN to the DS directory on your device and use bootloader to launch it.

Feathered-Serpent, 2022-11-05:
Alright, then I will close the issue, as the documentation has been changed to fit the correct order. Thanks so far for you help!

closed 2022-11-05 by Feathered-Serpent

Feathered-Serpent, 2022-11-27:
Just want to say another thanks. Put everything on the SD card just as told and with the bootloader CD it automatically booted DreamShell from the SD then. The GD Ripper now was able to get the name of the discs and I was able to rip my few games now without problems. Great project!