DC-SWAT/DreamShell #33: Main git fails to build

MastaG, 2024-07-30:
I'm building per instructions in the readme, but after setting up the SDK, DS seems to fail to compile:
```
kos-cc  -c src/mat3d.c -o src/mat3d.o
src/mat3d.c:42:28: fout: ‘PLX_MAT_COUNT’ undeclared here (not in a function)
   42 | static matrix_t trans_mats[PLX_MAT_COUNT] __attribute__((aligned(32)));
      |                            ^~~~~~~~~~~~~
src/mat3d.c: In functie ‘plx_mat3d_perspective’:
src/mat3d.c:120:9: let op: implicit declaration of function ‘plx_mat3d_frustum’; did you mean ‘plx_mat3d_store’? [-Wimplicit-function-declaration]
  120 |         plx_mat3d_frustum(xmin, xmax, ymin, ymax, znear, zfar);
      |         ^~~~~~~~~~~~~~~~~
      |         plx_mat3d_store
src/mat3d.c: Op bovenste niveau:
src/mat3d.c:130:6: let op: conflicting types for ‘plx_mat3d_frustum’; have ‘void(float,  float,  float,  float,  float,  float)’
  130 | void plx_mat3d_frustum(float left, float right,
      |      ^~~~~~~~~~~~~~~~~
src/mat3d.c:120:9: note: previous implicit declaration of ‘plx_mat3d_frustum’ with type ‘void(float,  float,  float,  float,  float,  float)’
  120 |         plx_mat3d_frustum(xmin, xmax, ymin, ymax, znear, zfar);
      |         ^~~~~~~~~~~~~~~~~
src/mat3d.c: In functie ‘plx_mat3d_push’:
src/mat3d.c:159:14: fout: ‘PLX_MAT_MODELVIEW’ undeclared (first use in this function)
  159 |         case PLX_MAT_MODELVIEW:
      |              ^~~~~~~~~~~~~~~~~
src/mat3d.c:159:14: note: each undeclared identifier is reported only once for each function it appears in
src/mat3d.c:168:14: fout: ‘PLX_MAT_PROJECTION’ undeclared (first use in this function)
  168 |         case PLX_MAT_PROJECTION:
      |              ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In functie ‘plx_mat3d_pop’:
src/mat3d.c:185:14: fout: ‘PLX_MAT_MODELVIEW’ undeclared (first use in this function)
  185 |         case PLX_MAT_MODELVIEW:
      |              ^~~~~~~~~~~~~~~~~
src/mat3d.c:194:14: fout: ‘PLX_MAT_PROJECTION’ undeclared (first use in this function)
  194 |         case PLX_MAT_PROJECTION:
      |              ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In functie ‘plx_mat3d_peek’:
src/mat3d.c:211:14: fout: ‘PLX_MAT_MODELVIEW’ undeclared (first use in this function)
  211 |         case PLX_MAT_MODELVIEW:
      |              ^~~~~~~~~~~~~~~~~
src/mat3d.c:219:14: fout: ‘PLX_MAT_PROJECTION’ undeclared (first use in this function)
  219 |         case PLX_MAT_PROJECTION:
      |              ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In functie ‘plx_mat3d_lookat’:
src/mat3d.c:373:9: let op: implicit declaration of function ‘mat_translate’; did you mean ‘mat_transform’? [-Wimplicit-function-declaration]
  373 |         mat_translate(-eye->x, -eye->y, -eye->z);
      |         ^~~~~~~~~~~~~
      |         mat_transform
src/mat3d.c: In functie ‘plx_mat3d_apply’:
src/mat3d.c:386:21: fout: ‘PLX_MAT_SCREENVIEW’ undeclared (first use in this function)
  386 |         if (mode != PLX_MAT_SCREENVIEW) {
      |                     ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In functie ‘plx_mat3d_apply_all’:
src/mat3d.c:420:32: fout: ‘PLX_MAT_PROJECTION’ undeclared (first use in this function)
  420 |         mat_apply(trans_mats + PLX_MAT_PROJECTION);
      |                                ^~~~~~~~~~~~~~~~~~
src/mat3d.c:421:32: fout: ‘PLX_MAT_WORLDVIEW’ undeclared (first use in this function)
  421 |         mat_apply(trans_mats + PLX_MAT_WORLDVIEW);
      |                                ^~~~~~~~~~~~~~~~~
src/mat3d.c:422:32: fout: ‘PLX_MAT_MODELVIEW’ undeclared (first use in this function)
  422 |         mat_apply(trans_mats + PLX_MAT_MODELVIEW);
      |                                ^~~~~~~~~~~~~~~~~
src/mat3d.c: In functie ‘plx_mat3d_init’:
src/mat3d.c:434:23: fout: ‘PLX_MAT_PROJECTION’ undeclared (first use in this function)
  434 |         matrix_mode = PLX_MAT_PROJECTION;
      |                       ^~~~~~~~~~~~~~~~~~
src/mat3d.c: Op bovenste niveau:
src/mat3d.c:42:17: let op: ‘trans_mats’ defined but not used [-Wunused-variable]
   42 | static matrix_t trans_mats[PLX_MAT_COUNT] __attribute__((aligned(32)));
      |                 ^~~~~~~~~~
make[2]: *** [/usr/local/dc/kos/kos/Makefile.rules:14: src/mat3d.o] Fout 1
make[2]: Map '/usr/local/dc/kos/kos/ds/lib/libparallax' wordt verlaten
make[1]: *** [Makefile:16: _dir_libparallax] Fout 2
make[1]: Map '/usr/local/dc/kos/kos/ds/lib' wordt verlaten
make: *** [Makefile:173: /usr/local/dc/kos/kos/ds/lib/libSDL_1.2.13.a] Fout 2
```

any ideas?

DC-SWAT, 2024-07-30:
I don't have this errors. Maybe you using old kos-ports version?

MastaG, 2024-07-30:
In the readme you mention: git checkout `cat ds/sdk/doc/KallistiOS.txt`
Perhaps this commit is too old?

MastaG, 2024-07-30:
I'll start from scratch and report back then

DC-SWAT, 2024-07-30:
Checkout latest commit. Seems I figured out.

MastaG, 2024-07-30:
Same happens:
```
make -C  libparallax
make[2]: Entering directory '/usr/local/dc/kos/kos/ds/lib/libparallax'
kos-cc  -c src/mat3d.c -o src/mat3d.o
src/mat3d.c:42:28: error: 'PLX_MAT_COUNT' undeclared here (not in a function)
   42 | static matrix_t trans_mats[PLX_MAT_COUNT] __attribute__((aligned(32)));
      |                            ^~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_perspective':
src/mat3d.c:120:9: warning: implicit declaration of function 'plx_mat3d_frustum'; did you mean 'plx_mat3d_store'? [-Wimplicit-function-declaration]
  120 |         plx_mat3d_frustum(xmin, xmax, ymin, ymax, znear, zfar);
      |         ^~~~~~~~~~~~~~~~~
      |         plx_mat3d_store
src/mat3d.c: At top level:
src/mat3d.c:130:6: warning: conflicting types for 'plx_mat3d_frustum'; have 'void(float,  float,  float,  float,  float,  float)'
  130 | void plx_mat3d_frustum(float left, float right,
      |      ^~~~~~~~~~~~~~~~~
src/mat3d.c:120:9: note: previous implicit declaration of 'plx_mat3d_frustum' with type 'void(float,  float,  float,  float,  float,  float)'
  120 |         plx_mat3d_frustum(xmin, xmax, ymin, ymax, znear, zfar);
      |         ^~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_push':
src/mat3d.c:159:14: error: 'PLX_MAT_MODELVIEW' undeclared (first use in this function)
  159 |         case PLX_MAT_MODELVIEW:
      |              ^~~~~~~~~~~~~~~~~
src/mat3d.c:159:14: note: each undeclared identifier is reported only once for each function it appears in
src/mat3d.c:168:14: error: 'PLX_MAT_PROJECTION' undeclared (first use in this function)
  168 |         case PLX_MAT_PROJECTION:
      |              ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_pop':
src/mat3d.c:185:14: error: 'PLX_MAT_MODELVIEW' undeclared (first use in this function)
  185 |         case PLX_MAT_MODELVIEW:
      |              ^~~~~~~~~~~~~~~~~
src/mat3d.c:194:14: error: 'PLX_MAT_PROJECTION' undeclared (first use in this function)
  194 |         case PLX_MAT_PROJECTION:
      |              ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_peek':
src/mat3d.c:211:14: error: 'PLX_MAT_MODELVIEW' undeclared (first use in this function)
  211 |         case PLX_MAT_MODELVIEW:
      |              ^~~~~~~~~~~~~~~~~
src/mat3d.c:219:14: error: 'PLX_MAT_PROJECTION' undeclared (first use in this function)
  219 |         case PLX_MAT_PROJECTION:
      |              ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_lookat':
src/mat3d.c:373:9: warning: implicit declaration of function 'mat_translate'; did you mean 'mat_transform'? [-Wimplicit-function-declaration]
  373 |         mat_translate(-eye->x, -eye->y, -eye->z);
      |         ^~~~~~~~~~~~~
      |         mat_transform
src/mat3d.c: In function 'plx_mat3d_apply':
src/mat3d.c:386:21: error: 'PLX_MAT_SCREENVIEW' undeclared (first use in this function)
  386 |         if (mode != PLX_MAT_SCREENVIEW) {
      |                     ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_apply_all':
src/mat3d.c:420:32: error: 'PLX_MAT_PROJECTION' undeclared (first use in this function)
  420 |         mat_apply(trans_mats + PLX_MAT_PROJECTION);
      |                                ^~~~~~~~~~~~~~~~~~
src/mat3d.c:421:32: error: 'PLX_MAT_WORLDVIEW' undeclared (first use in this function)
  421 |         mat_apply(trans_mats + PLX_MAT_WORLDVIEW);
      |                                ^~~~~~~~~~~~~~~~~
src/mat3d.c:422:32: error: 'PLX_MAT_MODELVIEW' undeclared (first use in this function)
  422 |         mat_apply(trans_mats + PLX_MAT_MODELVIEW);
      |                                ^~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_init':
src/mat3d.c:434:23: error: 'PLX_MAT_PROJECTION' undeclared (first use in this function)
  434 |         matrix_mode = PLX_MAT_PROJECTION;
      |                       ^~~~~~~~~~~~~~~~~~
src/mat3d.c: At top level:
src/mat3d.c:42:17: warning: 'trans_mats' defined but not used [-Wunused-variable]
   42 | static matrix_t trans_mats[PLX_MAT_COUNT] __attribute__((aligned(32)));
      |                 ^~~~~~~~~~
make[2]: *** [/usr/local/dc/kos/kos/Makefile.rules:14: src/mat3d.o] Error 1
make[2]: Leaving directory '/usr/local/dc/kos/kos/ds/lib/libparallax'
make[1]: *** [Makefile:16: _dir_libparallax] Error 2
make[1]: Leaving directory '/usr/local/dc/kos/kos/ds/lib'
make: *** [Makefile:173: /usr/local/dc/kos/kos/ds/lib/libSDL_1.2.13.a] Error 2
```

This is without doing:
```
git checkout `cat ds/sdk/doc/KallistiOS.txt`
```

DC-SWAT, 2024-07-30:
`git fetch && git checkout origin/master` in DS source directory.

MastaG, 2024-07-30:
I'm on the latest ds master:
```
mastag@razerblade15:/usr/local/dc/kos/kos/ds$ git log
commit f1399aa471f6fd1740582935ad92a8fa291860ee (HEAD -> master, origin/master, origin/HEAD)
Author: DC-SWAT <swat@211.ru>
Date:   Tue Jul 30 16:02:11 2024 +0700

    [core] Update parallax headers
```

Still the same issue:

```
make -C  libparallax
make[2]: Entering directory '/usr/local/dc/kos/kos/ds/lib/libparallax'
kos-cc  -c src/mat3d.c -o src/mat3d.o
src/mat3d.c:42:28: error: 'PLX_MAT_COUNT' undeclared here (not in a function)
   42 | static matrix_t trans_mats[PLX_MAT_COUNT] __attribute__((aligned(32)));
      |                            ^~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_perspective':
src/mat3d.c:120:9: warning: implicit declaration of function 'plx_mat3d_frustum'; did you mean 'plx_mat3d_store'? [-Wimplicit-function-declaration]
  120 |         plx_mat3d_frustum(xmin, xmax, ymin, ymax, znear, zfar);
      |         ^~~~~~~~~~~~~~~~~
      |         plx_mat3d_store
src/mat3d.c: At top level:
src/mat3d.c:130:6: warning: conflicting types for 'plx_mat3d_frustum'; have 'void(float,  float,  float,  float,  float,  float)'
  130 | void plx_mat3d_frustum(float left, float right,
      |      ^~~~~~~~~~~~~~~~~
src/mat3d.c:120:9: note: previous implicit declaration of 'plx_mat3d_frustum' with type 'void(float,  float,  float,  float,  float,  float)'
  120 |         plx_mat3d_frustum(xmin, xmax, ymin, ymax, znear, zfar);
      |         ^~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_push':
src/mat3d.c:159:14: error: 'PLX_MAT_MODELVIEW' undeclared (first use in this function)
  159 |         case PLX_MAT_MODELVIEW:
      |              ^~~~~~~~~~~~~~~~~
src/mat3d.c:159:14: note: each undeclared identifier is reported only once for each function it appears in
src/mat3d.c:168:14: error: 'PLX_MAT_PROJECTION' undeclared (first use in this function)
  168 |         case PLX_MAT_PROJECTION:
      |              ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_pop':
src/mat3d.c:185:14: error: 'PLX_MAT_MODELVIEW' undeclared (first use in this function)
  185 |         case PLX_MAT_MODELVIEW:
      |              ^~~~~~~~~~~~~~~~~
src/mat3d.c:194:14: error: 'PLX_MAT_PROJECTION' undeclared (first use in this function)
  194 |         case PLX_MAT_PROJECTION:
      |              ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_peek':
src/mat3d.c:211:14: error: 'PLX_MAT_MODELVIEW' undeclared (first use in this function)
  211 |         case PLX_MAT_MODELVIEW:
      |              ^~~~~~~~~~~~~~~~~
src/mat3d.c:219:14: error: 'PLX_MAT_PROJECTION' undeclared (first use in this function)
  219 |         case PLX_MAT_PROJECTION:
      |              ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_lookat':
src/mat3d.c:373:9: warning: implicit declaration of function 'mat_translate'; did you mean 'mat_transform'? [-Wimplicit-function-declaration]
  373 |         mat_translate(-eye->x, -eye->y, -eye->z);
      |         ^~~~~~~~~~~~~
      |         mat_transform
src/mat3d.c: In function 'plx_mat3d_apply':
src/mat3d.c:386:21: error: 'PLX_MAT_SCREENVIEW' undeclared (first use in this function)
  386 |         if (mode != PLX_MAT_SCREENVIEW) {
      |                     ^~~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_apply_all':
src/mat3d.c:420:32: error: 'PLX_MAT_PROJECTION' undeclared (first use in this function)
  420 |         mat_apply(trans_mats + PLX_MAT_PROJECTION);
      |                                ^~~~~~~~~~~~~~~~~~
src/mat3d.c:421:32: error: 'PLX_MAT_WORLDVIEW' undeclared (first use in this function)
  421 |         mat_apply(trans_mats + PLX_MAT_WORLDVIEW);
      |                                ^~~~~~~~~~~~~~~~~
src/mat3d.c:422:32: error: 'PLX_MAT_MODELVIEW' undeclared (first use in this function)
  422 |         mat_apply(trans_mats + PLX_MAT_MODELVIEW);
      |                                ^~~~~~~~~~~~~~~~~
src/mat3d.c: In function 'plx_mat3d_init':
src/mat3d.c:434:23: error: 'PLX_MAT_PROJECTION' undeclared (first use in this function)
  434 |         matrix_mode = PLX_MAT_PROJECTION;
      |                       ^~~~~~~~~~~~~~~~~~
src/mat3d.c: At top level:
src/mat3d.c:42:17: warning: 'trans_mats' defined but not used [-Wunused-variable]
   42 | static matrix_t trans_mats[PLX_MAT_COUNT] __attribute__((aligned(32)));
      |                 ^~~~~~~~~~
make[2]: *** [/usr/local/dc/kos/kos/Makefile.rules:14: src/mat3d.o] Error 1
make[2]: Leaving directory '/usr/local/dc/kos/kos/ds/lib/libparallax'
make[1]: *** [Makefile:16: _dir_libparallax] Error 2
make[1]: Leaving directory '/usr/local/dc/kos/kos/ds/lib'
make: *** [Makefile:173: /usr/local/dc/kos/kos/ds/lib/libSDL_1.2.13.a] Error 2
```

Perhaps it's because KOS SDK was built without:
```
git checkout `cat ds/sdk/doc/KallistiOS.txt`
```

I notice libparallax is in kos-ports, does it somehow link/include it?

DC-SWAT, 2024-07-30:
KallistiOS version doesn't affected on this issue. But of course you should use version hash from `ds/sdk/doc/KallistiOS.txt`.
DS using internal libparallax instead of kos-ports, I was just thinking about the conflict in the headers.
Looks like you lost lib/libparallax/include/matrix.h because in this file this define's.

MastaG, 2024-07-30:
The file seems to be present.
See the git log and contents of the file.

```
mastag@razerblade15:/usr/local/dc/kos/kos/ds$ git status
On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   lib/mxml/configure

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	lib/mxml/configure~

no changes added to commit (use "git add" and/or "git commit -a")
mastag@razerblade15:/usr/local/dc/kos/kos/ds$ git log
commit 21173d01c5ac276d021db1fcc66d4a95dcb1d916 (HEAD -> master, origin/master, origin/HEAD)
Author: MastaG <mastag@gmail.com>
Date:   Tue Jul 30 11:50:04 2024 +0200

    img4dc: buildfixes (#34)
    
    - realpath() function requires: -D_XOPEN_SOURCE=500
    - mds4ds requires <string.h>

commit f1399aa471f6fd1740582935ad92a8fa291860ee
Author: DC-SWAT <swat@211.ru>
Date:   Tue Jul 30 16:02:11 2024 +0700

    [core] Update parallax headers
...

mastag@razerblade15:/usr/local/dc/kos/kos/ds$ cat lib/libparallax/include/matrix.h 
/* Parallax for KallistiOS ##version##

   matrix.h

   Copyright (C) 2002 Megan Potter

*/

#ifndef __PARALLAX_MATRIX
#define __PARALLAX_MATRIX

#include <sys/cdefs.h>
__BEGIN_DECLS

/**
  \file Matrix primitives. Half of these are just wrappers for the DC arch
  matrix routines. The main purpose of this is to try to keep client code
  less architecture specific so we can potentially port later more easily.

  The second half are routines adapted from KGL, and are intended for 3D
  usage primarily, though they can also be used for 2D if you want to do
  that (rotate, translate, etc). Like KGL, these use degrees.
 */

#include <dc/matrix.h>
#include <dc/matrix3d.h>

/* dc/matrix.h -- basic matrix register operations. */
#define plx_mat_store		mat_store
#define plx_mat_load		mat_load
#define plx_mat_identity	mat_identity
#define plx_mat_apply		mat_apply
#define plx_mat_transform	mat_transform
#define plx_mat_tfip_3d		mat_trans_single
#define plx_mat_tfip_3dw	mat_trans_single4
#define plx_mat_tfip_2d		mat_trans_single3

/* dc/matrix3d.h -- some of these are still useful in 2D. All of
   the following work on the matrix registers. */
#define plx_mat_rotate_x	mat_rotate_x
#define plx_mat_rotate_y	mat_rotate_y
#define plx_mat_rotate_z	mat_rotate_z
#define plx_mat_rotate		mat_rotate
#define plx_mat_translate	mat_translate
#define plx_mat_scale		mat_scale

/* The 3D matrix operations, somewhat simplified from KGL. All of these use
   the matrix regs, but do not primarily keep their values in them. To get
   the values out into the matrix regs (and usable) you'll want to set
   everything up and then call plx_mat3d_apply(). */

/** Call before doing anything else, or after switching video
    modes to setup some basic parameters. */
void plx_mat3d_init();

/** Set which matrix we are working on */
void plx_mat3d_mode(int mode);

/* Constants for plx_mat3d_mode and plx_mat3d_apply */
#define PLX_MAT_PROJECTION	0	/** Projection (frustum, screenview) matrix */
#define PLX_MAT_MODELVIEW	1	/** Modelview (rotate, scale) matrix */
#define PLX_MAT_SCREENVIEW	2	/** Internal screen view matrix */
#define PLX_MAT_SCRATCH		3	/** Temp matrix for user usage */
#define PLX_MAT_WORLDVIEW	4	/** Optional camera/worldview matrix */
#define PLX_MAT_COUNT		5

/** Load an identity matrix */
void plx_mat3d_identity();

/** Load a raw matrix */
void plx_mat3d_load(matrix_t * src);

/** Save a raw matrix */
void plx_mat3d_store(matrix_t * src);

/** Setup viewport parameters */
void plx_mat3d_viewport(int x1, int y1, int width, int height);

/** Setup a perspective matrix */
void plx_mat3d_perspective(float angle, float aspect, float znear, float zfar);

/** Setup a frustum matrix */
void plx_mat3d_frustum(float left, float right, float bottom, float top, float znear, float zfar);

/** Push a matrix on the stack */
void plx_mat3d_push();

/** Pop a matrix from the stack and reload it */
void plx_mat3d_pop();

/** Reload a matrix from the top of the stack, but don't pop it */
void plx_mat3d_peek();

/** Rotation */
void plx_mat3d_rotate(float angle, float x, float y, float z);

/** Scaling */
void plx_mat3d_scale(float x, float y, float z);

/** Translation */
void plx_mat3d_translate(float x, float y, float z);

/** Do a camera "look at" */
void plx_mat3d_lookat(const point_t * eye, const point_t * center, const vector_t * up);

/** Apply a matrix from one of the matrix modes to the matrix regs */
void plx_mat3d_apply(int mode);

/** Manually apply a matrix */
void plx_mat3d_apply_mat(matrix_t * src);

/** Apply all the matrices for a normal 3D scene */
void plx_mat3d_apply_all();

__END_DECLS

#endif	/* __PARALLAX_MATRIX */
```

DC-SWAT, 2024-07-30:
I have no idea why this happen for you... This is something that is not obvious.

MastaG, 2024-07-31:
I found out, if I manually enter the libparallax directory, it seems to work.
```
cd lib/libparallax/
make
cd ../..
make
```
So I think it has something to do with the top Makefile inside the root of the "ds" directory.

MastaG, 2024-07-31:
Yes it seems that on my system, the build somehow fails if you do ```make libs```
Which according to the toplevel Makefile does this:
```
libs: $(LIB_DIR)/libSDL_$(SDL_VER).a

$(LIB_DIR)/libSDL_$(SDL_VER).a:
        cd $(LIB_DIR) && make
```

Somehow this causes libparallax mat3d.c to fail to find includes.
The workaround for me is to do it manually first.
```
cd lib && make
cd ..
make
```
Very strange.. somehow the toplevel Makefile messes things up.

MastaG, 2024-07-31:
Next issue, there seems to be no minilzo.so library being generated for linking.

```
mastag@razerblade15:/usr/local/dc/kos/kos/ds/modules$ LANG=C make
make -C  minilzo
make[1]: Entering directory '/usr/local/dc/kos/kos/ds/modules/minilzo'
rm -f minilzo.klf
rm -f libminilzo.a
/opt/toolchains/dc/sh-elf/bin/sh-elf-gcc -g -ml -m4-single-only -O2 -g -Wl,-Ttext=0x00000000 -e _start -nostartfiles -nodefaultlibs -o dbg-minilzo.klf module.o minilzo.o exports.o -L/usr/local/dc/kos/kos/lib/dreamcast -L/usr/local/dc/kos/kos/addons/lib/dreamcast -L/usr/local/dc/kos/kos/../kos-ports/lib -L/usr/local/dc/kos/kos/ds/sdk/lib -Wl,--start-group  -lds -lkallisti_exports -lds_gcc -Wl,--end-group
/opt/toolchains/dc/sh-elf/bin/sh-elf-gcc -g -ml -m4-single-only -O2 -Wl,-d -Wl,-r -Wl,-S -Wl,-x -nostartfiles -nodefaultlibs -o minilzo.klf -Wl,-T /usr/local/dc/kos/kos/ds/sdk/shlelf.xr module.o minilzo.o exports.o -L/usr/local/dc/kos/kos/lib/dreamcast -L/usr/local/dc/kos/kos/addons/lib/dreamcast -L/usr/local/dc/kos/kos/../kos-ports/lib -L/usr/local/dc/kos/kos/ds/sdk/lib -Wl,--start-group   -Wl,--end-group
\033[42m Compiled: minilzo v2.0.9 build 16 \033[0m
   text	   data	    bss	    dec	    hex	filename
   4449	    404	      0	   4853	   12f5	minilzo.klf
make[1]: Leaving directory '/usr/local/dc/kos/kos/ds/modules/minilzo'
make -C  isofs
make[1]: Entering directory '/usr/local/dc/kos/kos/ds/modules/isofs'
rm -f isofs.klf
rm -f libisofs.a
/opt/toolchains/dc/sh-elf/bin/sh-elf-gcc -g -ml -m4-single-only -O2 -g -Wl,-Ttext=0x00000000 -e _start -nostartfiles -nodefaultlibs -o dbg-isofs.klf module.o fs_iso9660.o map2dev.o ciso.o cdi.o gdi.o exports.o -L/usr/local/dc/kos/kos/lib/dreamcast -L/usr/local/dc/kos/kos/addons/lib/dreamcast -L/usr/local/dc/kos/kos/../kos-ports/lib -L/usr/local/dc/kos/kos/ds/sdk/lib -Wl,--start-group  -lds -lminilzo -lkallisti_exports -lds_gcc -Wl,--end-group
/opt/toolchains/dc/sh-elf/lib/gcc/sh-elf/13.2.0/../../../../sh-elf/bin/ld: cannot find -lminilzo: No such file or directory
collect2: error: ld returned 1 exit status
make[1]: *** [../../sdk/Makefile.loadable:116: isofs.klf] Error 1
make[1]: Leaving directory '/usr/local/dc/kos/kos/ds/modules/isofs'
make: *** [Makefile:18: _dir_isofs] Error 2
mastag@razerblade15:/usr/local/dc/kos/kos/ds/modules$ ls -ltr minilzo/
totaal 452
-rw-r--r--. 1 mastag mastag 208608 31 jul 14:36 minilzo.c
-rw-r--r--. 1 mastag mastag    282 31 jul 14:36 exports.txt
-rw-r--r--. 1 mastag mastag    185 31 jul 14:36 module.c
-rw-r--r--. 1 mastag mastag   9860 31 jul 16:28 module.o
-rw-r--r--. 1 mastag mastag  37028 31 jul 16:28 minilzo.o
-rw-r--r--. 1 mastag mastag   1073 31 jul 16:28 exports.c
-rw-r--r--. 1 mastag mastag   4652 31 jul 16:28 exports.o
-rw-r--r--. 1 mastag mastag    593 31 jul 16:35 Makefile
-rw-r--r--. 1 mastag mastag      3 31 jul 16:35 minilzo.build
-rwxr-xr-x. 1 mastag mastag 158444 31 jul 16:35 dbg-minilzo.klf
-rw-r--r--. 1 mastag mastag   9980 31 jul 16:35 minilzo.klf
```

It seems to link a minilzo.klf, but that is not what gcc seeks for linking isofs.klf.
The linker probably looks for ```minilzo.so```.

DC-SWAT, 2024-08-01:
`.klf` is `.so` for DS. In your case missed static library `.a`.
Look in `sdk/lib` for it.
What environment do you use? You catch issues that shouldn't exist.

DC-SWAT, 2024-08-01:
I updated KOS to the latest version and rebuilt everything and got the same issue with parallax library. I will fix it.

DC-SWAT, 2024-08-01:
I fixed issues with libparallax and modules. Try again from scratch please.

MastaG, 2024-08-01:
Thank you DC-SWAT :)
libparallax is fixed now.

However the modules still fail for me:
```
mastag@razerblade15:/usr/local/dc/kos/kos/ds/modules$ make
make -C  minilzo
make[1]: Map '/usr/local/dc/kos/kos/ds/modules/minilzo' wordt binnengegaan
rm -f minilzo.klf
cat: minilzo.build: No such file or directory
/bin/sh: line 1: 1: command not found
rm -f libminilzo.a
cat: minilzo.build: No such file or directory
/bin/sh: line 1: 1: command not found
kos-cc  -c module.c -o module.o
kos-cc  -c minilzo.c -o minilzo.o
/usr/local/dc/kos/kos/utils/genexports/genexports.sh exports.txt exports.c ds_minilzo_symtab
kos-cc  -c exports.c -o exports.o
/opt/toolchains/dc/sh-elf/bin/sh-elf-gcc -g -ml -m4-single-only -O2 -g -Wl,-Ttext=0x00000000 -e _start -nostartfiles -nodefaultlibs -o dbg-minilzo.klf module.o minilzo.o exports.o -L/usr/local/dc/kos/kos/lib/dreamcast -L/usr/local/dc/kos/kos/addons/lib/dreamcast -L/usr/local/dc/kos/kos/../kos-ports/lib -L/usr/local/dc/kos/kos/ds/sdk/lib -Wl,--start-group  -lds -lkallisti_exports -lds_gcc -Wl,--end-group
/opt/toolchains/dc/sh-elf/bin/sh-elf-gcc -g -ml -m4-single-only -O2 -Wl,-d -Wl,-r -Wl,-S -Wl,-x -nostartfiles -nodefaultlibs -o minilzo.klf -Wl,-T /usr/local/dc/kos/kos/ds/sdk/shlelf.xr module.o minilzo.o exports.o -L/usr/local/dc/kos/kos/lib/dreamcast -L/usr/local/dc/kos/kos/addons/lib/dreamcast -L/usr/local/dc/kos/kos/../kos-ports/lib -L/usr/local/dc/kos/kos/ds/sdk/lib -Wl,--start-group   -Wl,--end-group
\033[42m Compiled: minilzo v2.0.9 build 1 \033[0m
   text	   data	    bss	    dec	    hex	filename
   4449	    404	      0	   4853	   12f5	minilzo.klf
make[1]: Map '/usr/local/dc/kos/kos/ds/modules/minilzo' wordt verlaten
make -C  isofs
make[1]: Map '/usr/local/dc/kos/kos/ds/modules/isofs' wordt binnengegaan
rm -f isofs.klf
cat: isofs.build: No such file or directory
/bin/sh: line 1: 1: command not found
rm -f libisofs.a
cat: isofs.build: No such file or directory
/bin/sh: line 1: 1: command not found
kos-cc  -c module.c -o module.o
kos-cc  -c fs_iso9660.c -o fs_iso9660.o
kos-cc  -c map2dev.c -o map2dev.o
kos-cc  -c ciso.c -o ciso.o
kos-cc  -c cdi.c -o cdi.o
kos-cc  -c gdi.c -o gdi.o
/usr/local/dc/kos/kos/utils/genexports/genexports.sh exports.txt exports.c ds_isofs_symtab
kos-cc  -c exports.c -o exports.o
/opt/toolchains/dc/sh-elf/bin/sh-elf-gcc -g -ml -m4-single-only -O2 -g -Wl,-Ttext=0x00000000 -e _start -nostartfiles -nodefaultlibs -o dbg-isofs.klf module.o fs_iso9660.o map2dev.o ciso.o cdi.o gdi.o exports.o -L/usr/local/dc/kos/kos/lib/dreamcast -L/usr/local/dc/kos/kos/addons/lib/dreamcast -L/usr/local/dc/kos/kos/../kos-ports/lib -L/usr/local/dc/kos/kos/ds/sdk/lib -Wl,--start-group  -lds -lminilzo -lkallisti_exports -lds_gcc -Wl,--end-group
/opt/toolchains/dc/sh-elf/lib/gcc/sh-elf/13.2.0/../../../../sh-elf/bin/ld: cannot find -lminilzo: Bestand of map bestaat niet
collect2: fout: ld gaf exit-status 1 terug
make[1]: *** [../../sdk/Makefile.loadable:116: isofs.klf] Fout 1
make[1]: Map '/usr/local/dc/kos/kos/ds/modules/isofs' wordt verlaten
make: *** [Makefile:18: _dir_isofs] Fout 2
```

I'm using Fedora 40:
```
gcc-14.1.1-7.fc40.x86_64
gcc-c++-14.1.1-7.fc40.x86_64
make-4.4.1-6.fc40.x86_64
autoconf-2.71-10.fc40.noarch
```

DC-SWAT, 2024-08-01:
I figured out issue. Should be fixed now. Thanks for report!

MastaG, 2024-08-01:
No problem my friend :)

We're almost there now:

```
mastag@razerblade15:/usr/local/dc/kos/kos/ds/modules$ LANG=C make -j1
make -C  minilzo install
make[1]: Entering directory '/usr/local/dc/kos/kos/ds/modules/minilzo'
kos-cc  -c module.c -o module.o
cat: minilzo.build: No such file or directory
/bin/sh: line 1: 1: command not found
In file included from /usr/local/dc/kos/kos/ds/sdk/include/module.h:12,
                 from /usr/local/dc/kos/kos/ds/sdk/include/ds.h:17,
                 from module.c:7:
module.c: In function 'lib_get_version':
/usr/local/dc/kos/kos/ds/sdk/include/utils.h:36:51: error: expected expression before ')' token
   36 |                                         | (((BUILD) & 0xff) << 0) )
      |                                                   ^
/usr/local/dc/kos/kos/ds/sdk/include/module.h:91:24: note: in expansion of macro 'DS_MAKE_VER'
   91 |                 return DS_MAKE_VER(VER_MAJOR, VER_MINOR,                \
      |                        ^~~~~~~~~~~
/usr/local/dc/kos/kos/ds/sdk/include/module.h:97:9: note: in expansion of macro 'DEFAULT_MODULE_HEADER'
   97 |         DEFAULT_MODULE_HEADER(name);                                \
      |         ^~~~~~~~~~~~~~~~~~~~~
module.c:10:1: note: in expansion of macro 'DEFAULT_MODULE_EXPORTS'
   10 | DEFAULT_MODULE_EXPORTS(minilzo);
      | ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/dc/kos/kos/ds/sdk/include/module.h:93:9: warning: control reaches end of non-void function [-Wreturn-type]
   93 |         }
      |         ^
/usr/local/dc/kos/kos/ds/sdk/include/module.h:97:9: note: in expansion of macro 'DEFAULT_MODULE_HEADER'
   97 |         DEFAULT_MODULE_HEADER(name);                                \
      |         ^~~~~~~~~~~~~~~~~~~~~
module.c:10:1: note: in expansion of macro 'DEFAULT_MODULE_EXPORTS'
   10 | DEFAULT_MODULE_EXPORTS(minilzo);
      | ^~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [/usr/local/dc/kos/kos/Makefile.rules:14: module.o] Error 1
make[1]: Leaving directory '/usr/local/dc/kos/kos/ds/modules/minilzo'
make: *** [Makefile:18: _dir_minilzo] Error 2
```

DC-SWAT, 2024-08-01:
Fixed. Check it out.

MastaG, 2024-08-01:
Now it goes a lot further :)

```
...
make[1]: Leaving directory '/usr/local/dc/kos/kos/ds/modules/zip'
make -C  ini install
make[1]: Entering directory '/usr/local/dc/kos/kos/ds/modules/ini'
kos-cc  -c module.c -o module.o
/usr/local/dc/kos/kos/utils/genexports/genexports.sh exports.txt exports.c ds_ini_symtab
kos-cc  -c exports.c -o exports.o
/opt/toolchains/dc/sh-elf/bin/sh-elf-gcc -g -ml -m4-single-only -O2 -g -Wl,-Ttext=0x00000000 -e _start -nostartfiles -nodefaultlibs \
	-o dbg-ini.klf module.o exports.o -L/usr/local/dc/kos/kos/lib/dreamcast -L/usr/local/dc/kos/kos/addons/lib/dreamcast -L/usr/local/dc/kos/kos/../kos-ports/lib -L./libini/src/.libs -L/usr/local/dc/kos/kos/ds/sdk/lib -Wl,--start-group -lini -lds -lkallisti_exports -lds_gcc -Wl,--end-group
/opt/toolchains/dc/sh-elf/lib/gcc/sh-elf/13.2.0/../../../../sh-elf/bin/ld: cannot find -lini: No such file or directory
collect2: error: ld returned 1 exit status
make[1]: *** [../../sdk/Makefile.loadable:116: ini.klf] Error 1
make[1]: Leaving directory '/usr/local/dc/kos/kos/ds/modules/ini'
make: *** [Makefile:18: _dir_ini] Error 2
```

MastaG, 2024-08-01:
Yes!
All make commands passed now :)
Sorry to take up so much of your time.
Will copy to my CF and report back.

DC-SWAT, 2024-08-01:
You have nothing to apologize for, you helped me solve some issues. Thanks for testing!
Close this issue if you have success.

MastaG, 2024-08-01:
It boots and I was pleasantly greeted by a nice sound and new beautiful UI.
Amazing!
I think renaming wav modile wasnt fully completed.
It seems to be missing when running the iso loader.

[image]

DC-SWAT, 2024-08-01:
Yeah thanks, fixed too.

DC-SWAT, 2024-08-01:
Also the same for new Games Menu app.

MastaG, 2024-08-01:
Holy... this is amazing.. it even plays music in the isoloader menu!

You've been working really hard (and silently).
Can't wait for the next release.

All good now !

DC-SWAT, 2024-08-01:
This features (UI and music) presents in 4.0.0 Release: https://github.com/DC-SWAT/DreamShell/releases/tag/v4.0.0.Release
Currently I develop 4.0.1

MastaG, 2024-08-01:
Ah I must have missed the new UI then.
I did hit another problem though.
I was previously booting using `bios/ds/boot_loader_devkit.bios` image from a previous DreamShell release.

I've decided to flash `bios/ds/boot_loader_retail.bios` from the release I just compiled.
When turning on the console it now shows the retail boot swirl.
But after that it hangs on this screen:

[image]

It's not a problem, because my console features two flashable tsop chips with a switch.
So I can easily switch to the retail bios chip, boot from disc and flash again.

DC-SWAT, 2024-08-01:
Yes this is known issue. 
This happened during the next update in KOS and the problem is most likely in it. As soon as this is fixed, I will update.
Just use bootloader from release package yet.

closed 2024-08-01 by MastaG