DC-SWAT/DreamShell #37: [low-prio] Feature requests:

MastaG, 2024-08-01:
If possible I'd like to request two features for the next release.

- Option to automatically connect the Broadband Adapter and start FTP server during DS startup.
  This is the first step towards remote debugging.

- Controller hotplug support.
  I'm using a BlueRetro (bluetooth) adapter and when I connect (pair) controller after DS has been started, it will not work.
  The controller has to be connected before DS boots (during boot swirl).

DC-SWAT, 2024-08-01:
1. You can do it in startup.lua if you want. But connecting takes a time.
2. Hotplug supported for native controllers. This is KOS and/or blueretro issues, not DS.

DC-SWAT, 2024-08-02:
Uncomment `ftpd` module in `lua/startup.lua` and before line `OpenApp(os.getenv("APP"));` add this:
```
os.execute("net --init");
os.execute("ftpd -s -p 21 -d /");
```

And you will have a network and an FTP server when the DS starts. But the start itself will take longer.

MastaG, 2024-08-02:
Thanks a lot!

closed 2024-08-02 by MastaG