DC-SWAT/DreamShell #38: Iso loader not working in current master git

MastaG, 2024-08-01:
I've tried both Quake 3 Arena and Unreal Tournament in todays build.

- Quake 3 will freeze at black screen after the intro videos
- Unreal Tournament will reboot the console after into

Both are working fine with the same settings on 4.0.0 final release.

DC-SWAT, 2024-08-01:
I know. ISO loader in development process, this is a beta and not fully working yet. Use release version for general purpose, git for devs.

DC-SWAT, 2024-08-02:
To be precise, it almost works, but there are some issues due to the new functionality that I am still struggling with.
That's why version 0.8.1 is marked as Beta.

MastaG, 2024-08-02:
Alright, if you like you can close this bug.
Now that you've solved all issues when building from scratch, I can easily test it from time to time :)

closed 2024-08-02 by DC-SWAT