DC-SWAT/DreamShell #80: A question about you Triple Bios mod for the Dreamcast.

iVirtualPlays, 2024-12-23:
I don't really care about the triple bios mod for the Dreamcast, it's completely useless for me. But I'm super interested in the dual bios mod using a single MX29L3211 bios chip, where I can switch from Dreamshell/Retro Dream to the standard region free bios. This is way better than PCB's hanging off the bios chip, or the messy Piggyback bios mod which is also a pain to install. Could you make a guide for this? Or could you just sell those chips preinstalled with 2 bio's instead of 1. Having one bios chip which contains two bios's would make for a much cleaner install with better reliability.

DC-SWAT, 2024-12-24:
I don't sell anything and I don't have these chips. Also I don't recommend doing this, it's always better to leave the original BIOS, in case you flash something wrong, you can always boot from the GD-ROM or GDEmu to flash it again. It's important as a backup, which can't be accidentally damaged.

iVirtualPlays, 2024-12-24:
> I don't sell anything and I don't have these chips. Also I don't recommend doing this, it's always better to leave the original BIOS, in case you flash something wrong, you can always boot from the GD-ROM or GDEmu to flash it again. It's important as a backup, which can't be accidentally damaged.

I usually buy them preflashed and already have a Piggy Back Bios installed. Was just looking to switch over to have two bios's on a single chip that I'm never going to reflash. One to load Dreamshell and the other for the Region Free Bios. I'm not interested in ODE's since I prefer to play physical games along with switching over to my CF card when loading up Dreamshell.

DC-SWAT, 2025-01-02:
You can merge two BIOS into one with linux console command:
`cat first.bios second.bios > both.bios`

iVirtualPlays, 2025-01-03:
Would love to be able to do this. I never flashed chips before.

DC-SWAT, 2025-01-05:
> Would love to be able to do this. I never flashed chips before. Do I just drag and drop both bios files into the Command Prompt?

Nope. Just type from keyboard in command line :))

DC-SWAT, 2025-01-27:
I have improved BIOS Flasher application, now you can flash several firmwares into your chip. But you need to switch 2 MB banks on it, it is impossible to flash more at once, because there is not enough address space.

DC-SWAT, 2025-01-28:
> I see, I did order a 29L3211 chip preflashed. I just need to figure out how to install this chip into a Dreamcast. Do I just install it like a Piggyback Bios mod lifting up the same pins using 10K Resistors? And once I get it working by using the toggle switch. Can I flash a different bios over to each individual bank?

I think you can do that. But it's better to check with those who work with hardware mods. I'm more into software.
Maybe @megavolt85 can help with that.

megavolt85, 2025-01-29:
the mx29l3211 pinout is a little different, the /WE signal is on pin1 and pin44 is used to switch memory banks

iVirtualPlays, 2025-02-26:
Sorry for the delay, I'm going try installing this mod. This is the diagram, that I will be following. Do I need to lift Pin 12, 44 and or pin 1?[image]

megavolt85, 2025-02-26:
yes, all three pins need to be lifted

iVirtualPlays, 2025-02-26:
> yes, all three pins need to be lifted

Thanks for the quick reply, I will be trying this tomorrow.

iVirtualPlays, 2025-02-26:
> yes, all three pins need to be lifted

Only Pin 12 needs to be lifted since Pin 1 and 44 don't go anywhere. [image]

iVirtualPlays, 2025-02-27:
My first attempt and no luck at the moment, it just gives me a black screen. This unit has had failed Dragoncity PCB and Piggy Back Bios mods, hence the broken traces. I have repaired them but perhaps the repaired traces may have become detached during soldering. I got another unit, this time a stock Dreamcast to try this on with another spare I have. If that one works, then the traces will need to be looked at in this unit. If not then there is an issue with my install or the bios itself. 

[image]

[image]

[image]

[image]

iVirtualPlays, 2025-02-27:
I tried switching the wires around on the switch aswell as reflowing my install, but that doesn't fix the issue. It looks like the Pin 12 bank doesn't boot. But the Pin 44 bank boots just fine. I don't know where to go from here. I may need to just try to boot into Dreamshell and see if I can reflash both banks. [image]

[image]

iVirtualPlays, 2025-02-27:
[image]

Can't reflash the bios.

megavolt85, 2025-02-27:
the title of the topic indicates a triple BIOS mod, but in the end you install one chip and are surprised that it does not work.
it will not work with such a connection, all my advice was for a mod with two chips

iVirtualPlays, 2025-02-27:
Apologies for the confusion with the title. I wish I can edit that out. I was lead here by a Triple Bypass mod post, but I just wanted that one larger chip that was recommended for the Triple Bios mod to do a Dualboot bios install in place of the original bios. I want to find alternatives to the Piggyback Bios and the PCB add on mods, as I haven't had much success with those mods either. Using this larger 4mb makes for a much easier Install than other set ups, but there is no guide for installing it.

megavolt85, 2025-02-27:
[image]

iVirtualPlays, 2025-02-27:
> [image]

Thanks for the reply, much appreciated. Does that mean Pin 12 still stays cut? And Pin 44 gets wired to the nearest GND point like to Pin 32? And does the Pin 44 get wired up to the toggle switch?

megavolt85, 2025-02-27:
remove brown wire
disconnect green wire from pin12 and connect green wire to ground
remove resistor from pin12
solder pin12 to mainboard

megavolt85, 2025-02-27:
yes

iVirtualPlays, 2025-02-27:
Apologies for the confusion with the title. I wish I can edit that out. I was lead here by a Triple Bypass mod post, but I just wanted that one larger chip that was recommended for the Triple Bios mod to do a Dualboot bios install in place of the original bios. I want to find alternatives to the Piggyback Bios and the PCB add on mods, as I haven't had much success with those mods either. Using this larger 4mb makes for a much easier Install than other set ups, but there is no guide for installing it. 

And one last question for when it starts working, can I reflash these bios's again? Can Dreamshell reflash a bios chip like this on two separate banks?

megavolt85, 2025-02-27:
yep, but not tested

Hardconsole, 2025-02-27:
Hello,
I'm the seller of this preflashed 29L3211.
iVirtualPlays send me a private message to flash this chip, I have never test it before so I can't offer him any help to mount it on Dreamcast.
Originaly I don't offer any support because I don't have enough time to help all customer, but it send me lot message with wrong wiring.
I think the correct wiring is :

[image]

megavolt85, 2025-02-27:
no, your diagram is not correct

Hardconsole, 2025-02-28:
> no, your diagram is not correct

please explain us

Hardconsole, 2025-02-28:
@megavolt85 ?

[image]

megavolt85, 2025-02-28:
yes it will work, but it is not recommended to supply power to the address line, instead use a 10k pull-up resistor

[image]

iVirtualPlays, 2025-02-28:
[image]

It seems like I got it to work. When I toggle the switch, the consoles boot up, but it looks like it has the same bios in each of the banks. Unless I'm using the wrong type of switch. My toggle switch is an on/on switch, and it has an extra third leg. Is that the correct switch to use? Or am I supposed to use the On/Off toggle switch?

megavolt85, 2025-02-28:
It seems to me that you need to solder the green wire to the central contact of the switch.
Use the multimeter to check it, in the first position of the circuit breaker the blue and green wires must be closed, and in the second open

iVirtualPlays, 2025-02-28:
I see, so I just need to find the on and off legs for that switch so it can work like an on and off switch.

iVirtualPlays, 2025-02-28:
[image]

Like this? I'll give it a test to see if this works.

megavolt85, 2025-02-28:
yep, now test it

iVirtualPlays, 2025-02-28:
Whoa! It's working. Thank you for your help Megavolt85. This is amazing.

DC-SWAT, 2025-03-01:
> Whoa! It's working. Thank you for your help Megavolt85. This is amazing.

Can I close the issue? You can do it yourself also.

iVirtualPlays, 2025-03-01:
Yes, I have yet to test to see if I can reflash the bios through Dreamshell, is the MX29l3211 chip supported through the Bios flasher?

DC-SWAT, 2025-03-01:
> Yes, I have yet to test to see if I can reflash the bios through Dreamshell, is the MX29l3211 chip supported through the Bios flasher?

Yes, but you need latest version which not released yet. You can build it from source code or ask in discord.

iVirtualPlays, 2025-03-01:
I see, I think I will just wait for the latest version.

iVirtualPlays, 2025-03-02:
> You can merge two BIOS into one with console command:
> 
> Linux:
> `cat first.bios second.bios > both.bios`
> 
> Windows:
> `copy first.bios + second.bios both.bios /B`

How do I merge a bios file in bin format and the other in bios format into one? I want to merge Japanese Cake Devkit version and Dreamboot into one bios file. 

https://mega.nz/folder/4QwEEL5J#d8XOWfL0W77qxxpzWNR08w

DC-SWAT, 2025-03-02:
> How do I merge a bios file in bin format and the other in bios format into one? I want to merge Japanese Cake Devkit version and Dreamboot into one bios file. [...]

.bin and .bios it is the same format.

iVirtualPlays, 2025-03-02:
Oh ok, merging them doesn't seem to work on Windows 11.

DC-SWAT, 2025-03-02:
[app_bios_flasher.klf.zip](https://github.com/user-attachments/files/19043493/app_bios_flasher.klf.zip) - replace in DS/apps/bios_fkasher/modules and try.

DC-SWAT, 2025-03-02:
> Oh ok, merging them doesn't seem to work on Windows 11.

You don't need merging it. You need switch banks on your flash chip (if it more than 2MB) and flash every bios file separatelly.

iVirtualPlays, 2025-03-02:
> try

Okay I think I will try the Dreamshell method, but I think my other chip is a blank chip. So I'm going to need to flash that one on a PC.

iVirtualPlays, 2025-09-02:
> Did you find out everything you wanted? Can I close this issue @iVirtualPlays?

Yes indeed thanks to Megavolt85 I now have a 4mb dual bios mod installed in my DC: https://www.reddit.com/r/dreamcast/s/mDMCBdDPfJ

closed 2025-09-02 by DC-SWAT