DC-SWAT/DreamShell #85: Not able to create virtual vmu

sudo-lupus666, 2025-01-06:
First of all, thank you for the hardwork. My issue: I try creating virtual vmus, but either the game doesn't work after I attempt to use this resource through Dreamshell or the game does start, but with no support to the virtual vmu at all. What am I doing wrong?

I open Iso loader. Select a game, click on the puzzlish square and under emulation I chose either shared dump or private dumb. Save, try to execute the game. Either the game doesn't work or the virtual vmu is not available.

DC-SWAT, 2025-01-06:
Which game did you try? VMU emulation does not work in all games unfortunately.

sudo-lupus666, 2025-01-06:
Ohh, I see. I understand, now. I tried with Sonic Adventure, DCA3, Dino Crisis and Daytona. Actually, I got a vmu on Daytona, but the game said it was full and couldn't save on it.

sudo-lupus666, 2025-01-06:
Thank you! The vmu manager is more than enough for now. Keep up the good work.

closed 2025-01-08 by DC-SWAT