DC-SWAT/DreamShell #115: GD Play = No Disc ?

trev1976, 2025-07-13:
Hi 

I installed an aliexpress bios today which contains dreamshell , i then used a SD Card reader and installed the latest dreamshell and it boots fine. Problem is i want to use my retail discs for all regions JPN/PAL Mainly but when i select GD Play it just says i have no disc ? I also installed a switch so i could switch back to standard bios and when activated that allows me to boot my original PAL games (PAL Console). 

So yeah , im confused 

Please help. 

Thanks

megavolt85, 2025-07-13:
if noGDROM BIOS is flashed, then GDPlay and GDRiper applications will not work

megavolt85, 2025-07-13:
use BiosFlasher for write latest BIOS FW without string noGDROM in name

trev1976, 2025-07-13:
Would it be the top one here ? Thanks 

https://ibb.co/6Jsg8MF9

megavolt85, 2025-07-13:
boot_loader_devkit.bios - startup animation from devkit.
boot_loader_retail.bios - standart boot animation.

trev1976, 2025-07-13:
Shes working now(booted Virtual On JPN Fine) , also have that nice start up animation. 

Still wont boot Shenmue 2 Pal though , i mean this isnt a big deal as i can swap back to my Standard PAL bios for this , just thought it was a bit odd.

Thanks

closed 2025-07-13 by trev1976