DC-SWAT/DreamShell #116: Add an option to create a folder in the File Manager?

PlayerGameSK, 2025-08-20:
As the title says, would it be possible to add an option to create folders in the File Manager? Would be great for sorting VMU backups and such without having to either plug the SD card into my PC or use console commands to create the folders (which is pretty tedious). If this already exists and I just (somehow) missed it then I apologize, but I tried everything I could in the File Manager and didn't see such an option anywhere...

DC-SWAT, 2025-08-27:
> Also how is renaming files meant to work? Does it require a keyboard? Cause when I try with a controller, no keyboard is showing up on the screen... 

Press Start to show/hide virtual keyboard.

DC-SWAT, 2025-09-03:
Implemented. Expect it in the next release.

PlayerGameSK, 2025-09-20:
So about when can we expect the next release to drop?

DC-SWAT, 2025-09-20:
> So about when can we expect the next release to drop?

Hopefully this month.