DC-SWAT/DreamShell #117: It seems like Dreamshell doesn't have good game compatibility.

iVirtualPlays, 2025-09-01:
I have an IDE Modded Dreamcast where I load my games through a 256gb CF Card formatted as Fat32 with 64kb Cluster Size. I have the latest Retro Dream with the latest Dreamshell set up, with a collection GDI Dreamcast games with a mix of US and Pal games with a few NTSC J games loaded onto the CF Card. And besides all of this, compatibility is hit and miss, some games like Sonic Adventure, Sega GT and Daytona USA load up perfectly fine with faster load times thanks to the IDE Mod with the CF Card set up. But other games like Shenmue and Sega Rally 2 doesn't boot up at all.

I really wanted to have an ODE like setup whilst retaining the GDRom Drive, but it seems like Dreamshell just isn't designed to run games like an ODE software would. I also went over the Dreamshell compatibility chart for setting up the games according to the compatibility chart, but Shenmue doesn't even boot up, and other games suffer from similar issues. What's causing this? I already reformatted my CF Card and copied everything over in one go, but it made no difference. And I'm not using those god awful serial adaptors to load games from. 

My VA1 Dreamcast specs tested in full working order: https://www.reddit.com/r/dreamcast/s/yuGyVvHY9E

4mb Dualboot Bios loaded with Dreamboot and Japanese Cake. 

Recapped with Polymer Capacitors 

Dragon City IDE mod PCB

IDE to CF Adaptor 

256gb CF Card

Noctua 40mm 5v Fan

ReDream Pico PSU

New Arctic Thermal Pads 1.5mm

iVirtualPlays, 2025-09-01:
[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

DC-SWAT, 2025-09-02:
1. The last official RetroDream update was in 2023 (and the necessary updates are generally in 2021) and it does not support the new parameters of the 0.8+ loaders. At the request of Dragoncity, I compiled new loaders specifically with hardcoded parameters. How you use them there, I have no idea, because each game needs presets. I told him a long time ago to stop doing this and use DreamShell directly, but he wants to give a choice, so he let you choose this.

2. DreamShell has full compatibility with games, in some cases better than ODE, in some a little worse, but in general you can play everything. To make sure of this, first use DreamShell version 4.0.1 or 4.0.2 or newer with ISO Loader or Games Menu applications. Version 4.0.2 have built-in presets for all games, but perhaps for some images you need to set the settings manually, which can be viewed in this compatibility list:
https://drive.google.com/file/d/1Xd-Wh-rSrlyBEXhQkovPBIxzm4fLB744/view?usp=drive_link
All the games you listed work fine, I personally played Shenmue recently. Except for Sega Rally 2, it only works with a specific image, I don't remember which one, ask on the forum or discord. Oh, and there is no CDDA in this game yet. This is the first WinCE game, it has a lot of deviations, it needs to be worked on more closely, but I don’t want to waste time on it yet. Honestly, I don't really like this one, V-Rally 2 is much better for me.

3. DreamShell supports images in formats: ISO, GDI, CDI, CSO and even custom GDIs that are not supported by emulators/ODE.
Better use ISO or optimized GDI (with .iso tracks), for some games need defragmented images on filesystem.
CDI and non-optimzied GDI works worse, so better use ISO and GDI-opt. You can download every game in this formats today, it's not an issue.
Your problem with the GTA3 most likely lies on the surface, you probably needed to select Homebrew in the settings simply, since some Homebrew images use an encrypted binary and can't be auto detected.

4. You have a great setup, just don't chase RetroDream, use ISO and optimized GDI images, look for compatibility list in some cases and you will be fine.

DC-SWAT, 2025-09-02:
> Thanks for the response, I will be sure to post about this issue on RetroDream GitHub. I will try loading up these games directly with Dreamshell with the new parmeter settings, first to see if that fixes this issue. I need to find an Optimisation App for these games, preferably one that can optimise more than a few dozen games since I have over 150 games to Optimise. Sega Rally 2 is just one of many games that isn't working through RetroDream with the old Paremters chart.

You can download ready to use images from pinned threads here:
http://www.dc-swat.ru/forum/forum-11.html
I doubt you play 150 games at once, download the ones you really need.

For batch optimizing of your images you can use this tool, but you need to compile it:
https://github.com/megavolt85/tools_for_DC/tree/master/gdiopt%20batch