DC-SWAT/DreamShell #118: Games not booting, the fault continues to persist.

iVirtualPlays, 2025-09-02:
In response to prior issue, where I was able to get Shenmue working. Some of the other games continue to not work, Sega Rally 2 doesn't boot, it gets stuck at executing, F355 Challenge boots to the menus, the Aklaim logo looks a bit messed up, and then it boots to a white screen. MSR Racing crashes at the menu,  GTA III has the same issue as Sega Rally 2, it gets stuck at executing. Is Dreamshell Compatible with CF cards. I went through the settings chart, to make sure it was set right. I essentially have an IDE to a 256gb CF card set up, but I don't why some of these game continue to not work. I will try GDi optimising some of these games, but then there is GTA 3 that is in CDi format.

DC-SWAT, 2025-09-03:
I use CF card for development and test, so I'm sure it works fine. This is the best choice for DreamShell.

Just use optmized GDI and ISO for all games. Also for some games need defragmented images on fs, MSR is this game.

DCA3 build system supported build in ISO format for DreamShell too, why you download CDI I don't know, get my images:

https://discord.com/channels/570500506972192768/1081861126620459038/

DC-SWAT, 2025-09-03:
Sega Rally 2 is the only problematic WinCE game, it works only `NTSC J` as I remember and without `CDDA` emulation. It is not worth focusing on it. I haven't dealt with such specific things yet, these problems are specific to this game only. Other games that you mentioned should work great.

iVirtualPlays, 2025-09-03:
I used GDi Patcher: https://segaxtreme.net/resources/dreamcast-gdi-quick-patcher.355/

GDi Optimizer doesn't work with my GDi games, also some of these games have raw tracks. I will try replacing these games and will replace GTA 3 with the ISO version. I will also reflow my IDE mod and adapter. The IDE CF Adaptor has Lead Free solder on it, so I will clean up the solder and reflow it with Lead solder.

DC-SWAT, 2025-09-03:
> I used GDi Patcher: https://segaxtreme.net/resources/dreamcast-gdi-quick-patcher.355/

This utility absolutely not needed for DreamShell ISO Loader, this is region patching utility, not optimizing.
DreamShell bypasses all these restrictions itself. You're more likely to ruin something by doing this than to do something useful by it.

> GDi Optimizer doesn't work with my GDi games, also some of these games have raw tracks. I will try replacing these games and will replace GTA 3 with the ISO version. I will also reflow my IDE mod [...]

Why doesn't it work? RAW tracks should remain as RAW tracks, because they are audio tracks. GDI optimizer only converts `.bin` tracks to `.iso` and that's it.

DC-SWAT, 2025-09-03:
By the way, I just tried Sega Rally 2 on the current development version of the ISO loader and it worked for me even with CDDA. So I hope you won't have problems with it in the new release.

iVirtualPlays, 2025-09-03:
Yeah I couldn't figure out how to GDi Optimise the GDi games, it doesn't work for me, perhaps it's not compatible with Windows 7. So I just downloaded the ones from your link. I guess Sega Rally 2 is unplayable on Dreamshell. And I can't seem to find GTA 3 in ISO format, it only comes in CDi format. Does that need to be converted? 

Btw I reflowed the IDE mod and the adaptor, and reformated the CF Card. Just waiting for everything to transfer over with those games replaced with the ones from the link you sent. I will see if it works. 

[image]

[image]

[image]

[image]

[image]

[image]

iVirtualPlays, 2025-09-03:
Another update looks like it's working. It was indeed the games themselves. Downloading it from your link got them running. Sega Rally 2 is working.

DC-SWAT, 2025-09-03:
> And I can't seem to find GTA 3 in ISO format, it only comes in CDi format. Does that need to be converted?

I gave you links where you can download. Open the spoiler by following the link on the forum:
http://www.dc-swat.ru/forum/thread-4146.html

Or go to Discord: 
https://discord.com/channels/570500506972192768/1081861126620459038/

> Another update looks like it's working. It was indeed the games themselves. Downloading it from your link instead of Archive.org got them running. Sega Rally 2 is working but some of the sound is missing as expected due to the CDDA issue.

Yeah, in next release CDDA will work in Sega Rally 2.

iVirtualPlays, 2025-09-03:
Thanks will download that, didn't know it was also on that thread. Is there a way to download all of them? I don't see any torrents, and I think i might have a whole bunch of unoptimised DC Games.

DC-SWAT, 2025-09-04:
> Thanks will download that, didn't know it was also on that thread. Is there a way to download all of them? I don't see any torrents, and I think i might have a whole bunch of unoptimised DC Games. Also is it possible to integrate GDi Optimiser into Dreamshell? If that's possible it would make things a lot easier.

There is no torrent.
Optimization on Dreamcast will take a lot of time, so I don't think it will be really a lot easier.

iVirtualPlays, 2025-09-04:
Thanks looks like it's resolved, already downloaded 80 games... Cough I meant I backed up 80 games that I legally own.

closed 2025-09-04 by DC-SWAT